Neotron-Compute/Neotron-Pico #2: Add Schotky to VSYS input as per Datasheet

thejpster, 2021-05-02:
https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf Section 4.5

[image]

IGBC, 2021-05-03:
if the intention of this is to switch power sources, there are better circuits that don't introduce a diode drop

thejpster, 2021-05-03:
The intention was to not blow anything up if someone plugs the micro-USB AB port into a PC or a phone charger whilst the Neotron Pico is turned on. The Schotky drop should be about 0.4V, which at the ~100mA the Pico uses is about 40mW.

The datasheet goes on to explain that a P-channel MOSFET, such as the Diodes DMG2305UX, will do the job and drop less voltage. Would you suggest we do that, or did you have another idea?

IGBC, 2021-05-07:
yea OK looks good to me.

closed 2021-05-25 by thejpster