Neotron-Compute/Neotron-Pico #50: 12bit VGA DAC non linear

DavePallant42, 2021-07-14:
One of the reasons the VGA DACs might be non linear going from 0111 to 1000 is that the digital outputs going into the R-2R networks are not perfect sources either because they do not go fully to 0V and 3.3V and also because they may have output impedance. Can be checked by measuring the voltages on the 12 outputs when set to 0111 0111 0111 and then 1000 1000 1000 and seeing how imperfect they are.

jimurai, 2021-07-31:
It might be worth do the same check as @DavePallant42 recommends but with the potential divider (attenuator to 0.7V) decoupled from the R-2R ladder (i.e. remove R509 and measure voltage at joint of R507 & R508). I haven't done the sums but input resistance of 15kΩ for potential divider feels like it could easily bias the most significant bit's 24kΩ path.

thejpster, 2022-07-29:
Reworked the DAC to be more compact, and use lower value resistors.

closed 2022-07-29 by thejpster