Neotron-Compute/Neotron-Pico #152: Replace the STM32F0 with a RP2040 and small Flash

thejpster, 2022-11-28:
Apparently the trick to soldering down an RP2040 by hand is to put a big via under the ground pad so you can fill it with solder from the underside. You then just flow each pad of the QFN like you would the leg of an SSOP. Or you let JLCPCB do it for you. Either way, it's cheaper than an STM32F0 and more powerful and has more IO.

thejpster, 2022-11-28:
The 2040 could possibly replace the MCP23S17 as well, which would be nice.

thejpster, 2022-12-03:
Duh, I forgot the RP2040 is *not* 5V tolerant and requires bi-directional buffers for the PS/2 port (or two resistors to drop the input level, and a low-side FET to pull the lines low for the output).

NollKollTroll, 2022-12-11:
Many ps/2 keyboards are speced to work at 3.3V, my own tests confirm this works with rp2040. But of course not all keyboards...

thejpster, 2022-12-17:
Hmm, not sure the RP2040 has enough pins to replace the MCP23S17. It only has 30, and we need 23 for existing BMC functions, plus another 4 to deal with the PS2 pins not being 5V tolerant (separate inputs and outputs). That only leaves a further 3 spare.

NollKollTroll, 2022-12-18:
Before using a 3.3V compatible keyboard I used a cheap pre-made PCB with BSS138 FETs that worked bi-directional as advertised. Something like this: https://www.adafruit.com/product/757

thejpster, 2022-12-18:
Hmm, that's a good idea actually. I should buy one and try it out.

thejpster, 2022-12-18:
Relevant NXP App Note linked to by Adafruit: https://cdn-shop.adafruit.com/datasheets/AN10441.pdf

thejpster, 2022-12-18:
So going back to the note above, if we can do external bi-directional level shifting, then an RP2040 has 7 extra pins. This can do 3 Chip Select lines (into the 74HC138 3:8 decoder) and four external interrupts for the four on-board slots (Slot 0 is the BMC itself, so that's internal, and Slot 1 is the SD Card and it doesn't need an interrupt). Or you'd run the interrupts into a priority 8:3 encoder, or a simple shift register like a 74HC165. If you can ditch the MCP23S17 and the STM32F0, then an RP2040 makes sense.

thejpster, 2022-12-18:
Of course, if we think a QFN is hand-solderable, you could switch to the STM32F030C8T6, which is an LQFP-48. Then you get about 16 more GPIO than the STM32F030K6T6 (LQFP-32) which is plenty to do the IRQ and CS pins; plus the firmware requires far fewer changes because it's largely the same as the one we have.

I'm going to close this.

closed 2022-12-18 by thejpster