Neotron-Compute/Neotron-Pico #154: BTX-style variant

erkinalp, 2022-12-02:
There were many office PCs in the BTX and microBTX form factors, that have their motherboards failed since. As BTX has the layout reversed, it is incompatible with ATX layout boards. Hence, add a microBTX variant with identical IO capabilities.

thejpster, 2022-12-03:
So this raises an interesting question. How does KiCAD handle a design with one schematic but two PCB layouts? Or does a BTX version require schematic changes?

The best way to handle this probably to fork the repo and create a "Neotron Pico BTX". If you have success and building one and can demonstrate it mechanically fits in a BTX case, we can look at moving that repository into this organisation for ongoing maintenance.

erkinalp, 2022-12-03:
A BTX version would not require schematic or component selection changes, but would require layout changes.

erkinalp, 2022-12-04:
@thejpster assign to me pls

thejpster, 2023-04-30:
Any progress with this?

erkinalp, 2023-04-30:
Not yet, but I will eventually deal with this.