Neotron-Compute/Neotron-Pico #159: probe-run instructions could use some improvement

9names, 2022-12-18:
#158 included the following note:

> **NOTE**: Linux Mint requires `open-ocd`, `libudev-dev` and `libusb-1.0-0-dev` before installing `probe-run`.

This is confusing to me, because probe-run does not use openocd. Also, `open-ocd` isn't the package name, so anyone trying to follow this advice will run into difficulty

It also includes this section on udev rules:

> For Linux Mint (and likely most other modern Linux distributions), the udev-rules also need to be updated for the SWD debug probe. The file `/etc/udev/rules.d/99-cmsis-dap.rules` should be created, containing:
> 
> ```text
> # cafe:4005 RP2040 Pico with CMSIS-DAP
> SUBSYSTEM=="usb", ATTR{idVendor}=="cafe", ATTR{idProduct}=="4005", MODE:="0666"
> ```
> 
> **NOTE:** Edit idVendor & idProduct to match VID & PID of the device you are using.
> 

probe-rs distribute a udev rules that covers all supported probes.
This file has one line that covers all CMSIS-DAP probes that have "CMSIS-DAP" in the USB product name, which is a requirement for a CMSIS-DAP compliant probe. this would be much more user-friendly than getting people to manually look-up and enter their probe's VID/PID pair.
https://probe.rs/docs/getting-started/probe-setup/

This could be an alternative to the above instructions:
```
wget https://probe.rs/files/69-probe-rs.rules
sudo mv 69-probe-rs.rules /etc/udev/rules.d/
udevadm control --reload
udevadm trigger
```

NollKollTroll, 2022-12-18:
I have no previous understanding of how Rust development works, or the requirements behind the different tools. With a fairly new install of Linux Mint 21 I could not use probe-run until I installed open-ocd. I have no idea why, or what parts were actually missing, but this was my experience. What is the proper name of the package?
That is a much better way to inform about how to get udev rules working, I wish I had found that info when doing the install :)

9names, 2022-12-18:
The package name is `openocd`.
So if you wanted to install it on Linux Mint using the command-line tools, you would run
```system
sudo apt install openocd
```

thejpster, 2022-12-18:
@9names ... any chance of a PR?

PR #161 (2022-12-20): Update probe-run setup instructions
Resolves #159
closed 2022-12-20 by thejpster