Neotron-Compute/Neotron-Pico #168: WASM-4 Support

jonititan, 2023-05-03:
Perhaps a useful future feature would be support for WASM-4 "carts"?
https://wasm4.org/docs/

It targets retrogaming and can be run in small devices as well as online.
- 4 color 160x160 pixel display
- 64 KB of memory
- 2 button gamepad

thejpster, 2023-05-03:
Interesting! I'll check it out, thanks.

jonititan, 2023-05-03:
Apparently some have ported it to small devices already.
https://twitter.com/alvaroviebrantz/status/1518343016011943939?s=20

In particular using the wasm3_engine including to run things on the pi Pico.

https://twitter.com/wasm3_engine/status/1360505531559985152?s=20

Might not be the best way for the Neotron-Pico but shows the hardware
itself can cope.

thejpster, 2023-05-07:
When the OS gains loadable applications I suggest this is written as a loadable application. There's probably nothing we need to change in the design of the Neotron Pico to support this though.

jonititan, 2023-05-07:
I think once that point is reached it would be a definite enhancement and
would make the Neotron even more attractive as a retro computing target.

The one issue being the multiplayer support of WASM-4.
Your previous comment on it not being a good idea to connect the Neotron to
the web is quite reasonable but does limit multiplayer to local lan.
However in that instance I'm not sure if WASM-4 can make the connections.
It relying on WebRTC and using WASM-4 servers for STUN & TURN they may not
have a local mode anticipated.
In their dev tutorials you run a server which allows local network devices
to join.(cool qr code joining method)
So it may be a non issue.

jonititan, 2023-05-07:
https://wasm4.org/docs/guides/multiplayer/#netplay